subutai-attic/liquid-router #9: Basic signal transfer between Zynq PL and Artix on LR.

metinburak, 2017-01-19:
We should send a signal from PL of Zynq using Linux uio interface or whatever suits, then we should read the signal from Artix 7 on LR and forward it to either emmc connector and observe by scope, or send it to another pin of Zynq PL and read it from Linux or whatever suits this application. With this test, we will be ready for LVDS applications.

Bekzade, 2017-01-19:
Tests with loopback signals passed. 5 input wires was connected with 5 output wires in the Artix7 FPGA. From Zynq PL we sent data to output 5 wire bus which is connected to input wires of Artix and receive the same data from input wires in Zynq PL.

Bekzade, 2017-01-20:
Block Design of test project in Zynq PL:
[image]

Block Design of test project in Artix FPGA
[image]

Waveform captured from Hardware Manager shows output and input wires. There is delay for 1 clock cycle  between output and input signals.
[image]

This signal loop can be checked by SW also. The **_axi_gpio_** IP core allows read status of input signal from `base_addr+8` offset address:
```
root@linaro-developer:/home/linaro/gpio_dev_mem# ./gpio-dev-mem-test -g 0x43c00000 -o 0
GPIO access through /dev/mem.
root@linaro-developer:/home/linaro/gpio_dev_mem# ./gpio-dev-mem-test -g 0x43c00000 -o 31
GPIO access through /dev/mem.
root@linaro-developer:/home/linaro/gpio_dev_mem# ./gpio-dev-mem-test -g 0x43c00008 -i
GPIO access through /dev/mem.
gpio dev-mem test: input: 0000001f
root@linaro-developer:/home/linaro/gpio_dev_mem# ./gpio-dev-mem-test -g 0x43c00000 -o 1
GPIO access through /dev/mem.
root@linaro-developer:/home/linaro/gpio_dev_mem# ./gpio-dev-mem-test -g 0x43c00008 -i
GPIO access through /dev/mem.
gpio dev-mem test: input: 00000001
root@linaro-developer:/home/linaro/gpio_dev_mem# 
```

closed 2017-01-22 by Bekzade